Parallel-7/FlashForgeUI-Electron #7: Camera not recognised on AD5X

hcoohb, 2025-09-25:
Hi,

I have added the official Camera to my AD5X and I confirmed that it works using the FlashMaker android App.
However, when I connect to FlashForge UI 1.0, it seems it does not recognised the camera and I get the following error when trying to view the preview: 
[image]
(on linux but I also checked on windows and it is the same)

There does not seem to be much in the logs, even with "debug mode":
```
[11:21:19 am] UI initialization complete - awaiting component system
[11:21:19 am] Component system initialized: 9 components
[11:21:19 am] Component system initialized: 9 components
[11:21:19 am] Renderer ready signal sent successfully
[11:21:23 am] Backend ready for AD5X (ad5x)
[11:21:23 am] Starting WebUI server for AD5X...
[11:21:23 am] WebUI server started successfully - remote access now available
```

All the other details on the printer status seems to be queried correctly

Let me know how I can help more

hcoohb, 2025-10-01:
I have seen in your reply of the other issue, that you too have a camera. Is it the flashforge Camera?
Do I need a specific setting so it loads in flashforge-UI?

GhostTypes, 2025-10-02:
Hello! Sorry to answer this so late!

[image]

You'll want to enable the Custom Camera setting, and restart the program so it fully applies. From then onwards, the camera should work normally! 

The program enables/disables the camera functionality based on the printer type, so by default it's only enabled for the 5M pro. This setting tells the program that you have a camera installed , and re-enables functionality.