Parallel-7/FlashForgeUI-Electron #8: File format for upload job on AD5X?

hcoohb, 2025-09-25:
Hi,

Sorry, another question
When using orcaslicer (not the flashforge version) to slice a file, I tried to upload it as a job using flashforge UI, however I get errors:

- When trying to upload a .gcode sliced file:
  [image]  
  So it seems the AD5X cannot receive a gcode file through flashforge (even though we can send it through to the AD5X straight from orcaslicer..)
  
  

- But what I really wanted to do was to hopefully export some sliced 3mf so I could use the IFS dialogue at the printer to select what spool number for what material, Unfortunately, it does not seem to work either.  
  I can select a sliced 3mf (.gcode.3mf) and the file is read just fine:
  [image]

  But when actually uploading to the printer it fails, with the message:
  [image]

  There is nothing new in the log however,

Is there a way to give you more info on the errors?

Using Flashforge UI 1.0 on linux

hcoohb, 2025-10-01:
Thanks for looking into this.

The reason I am not using orca-flashforge is that it is not available for linux systems and that is all I have at home...
Using regular OrcaSlicer, when uploading the gcode if starting the print without enabling IFS, the number of the filaments used by the printer will match what is in the gcode but it will also assume that filament number one is fully loader in the toolhead.
If enabling IFS at the printer, no need for any filament to be loaded at the toolhead, and the order of filament used are the same as placed in the slicer. 

BUT

with regular orcaslicer, we can also slice and export a sliced 3mf by doing "File>Export>Export plate sliced file". Doing so export a *.gcode.3mf. If I take that file on a USB and load on the printer it recognised the sliced 3mf and ask for filament selection at the printer screen. So I now the AD5X can read that file properly. I haven't found a way to use orcaslicer to send it directly. But I was hoping that  FlashforgeUI could as I saw it could load the file and recognised it, but then had the "Upload Failed" error.

So if that is possible, it would be awesome, as that mean I could send sliced 3mf generated with orcaslicer to the printer wihtout USB and still have the filament selection option.

GhostTypes, 2025-10-02:
No problem! Thank you for taking the time to explain everything. I totally forgot that option existed in the FIle menu 😅 

l tested with OrcaSlicer, and although it is picking up all the metadata for the .gcode.3mf file, for some reason it's not triggering the material selection dialog
[image]

I should be able to get this fixed pretty soon and have it work so you can select the materials and upload directly from here!

GhostTypes, 2025-10-04:
Hello!

I have tested with a file sliced from OrcaSlicer on [this](https://github.com/Parallel-7/FlashForgeUI-Electron/releases/tag/v1.0.1-7) latest alpha release, and it's working as expected on my end! 

I was able to select a file, the materials, and then it uploaded and started correctly. 

If you still encounter the same issue on this version, please:
- Run the app from CLI to get the console logs 
- Trigger the error, and press CTRL + Shift + I to open the electron debug menu
- Navigate to the console tab, right click on empty space to open the context menu, and select 'Copy Console' 

I'll be able to see exactly what is causing the error for you from these logs

closed 2025-10-16 by GhostTypes

hcoohb, 2025-10-16:
Sorry I am away from the printer for some time now, I could not test it...
Thanks for working on it!!